mkuf/prind #117: ustreamer not working?

nelsongraca, 2024-02-27:
been using this for quite some time, but ustreamer has stopped working.

    Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: exec: "/opt/ustreamer/ustreamer": stat /opt/ustreamer/ustreamer: no such file or directory: unknown

mkuf, 2024-02-27:
There seems to have been a change upstream which is causing issues starting at version `v5.52`.  
I removed those images from the registry and tagged a previous release w/ `latest` until this is resolved.  

Pulling the latest image now should clear the error. 

-Markus

ginkel, 2024-02-29:
Is it possible that reverting latest broke arm64? According to Docker Hub the latest tag only supports amd64.

mkuf, 2024-02-29:
Yes, that was an oversight from my side. 
I pulled the last working image on my linux/amd64 machine, tagged it as latest and pushed it again.  

I'll look into it.

mkuf, 2024-02-29:
Missing architectures have been copied, `latest` is now available for all platforms again.

closed 2024-02-29 by mkuf