mkuf/prind #137: moonraker-telegram-bot config in docker-compose.yaml needs update

mthqwork, 2024-03-14:
I have tried to update the stack today, and faced with a nice "no matching manifest for linux/arm/v7 in the manifest list entries" error message.
I have pinned down that lefskiy have removed all his/her repository from public, so docker can't pull it.

mkuf, 2024-03-14:
It does seem to have been removed w/ https://github.com/nlef/moonraker-telegram-bot/commit/3f5cbab15b0b3dacfe61619938ee70033abd4a03 in the development branch, maybe @nlef can shed some light on that?

The `latest` image still supports arm/v7 though, while the most recent build from development does not: 
[image]

mkuf, 2024-03-14:
Which tag did you try to pull @mthqwork?

mthqwork, 2024-03-14:
It was the default which came with the config, iirc it was "lefskiy/moonraker-telegram-bot:development"

mkuf, 2024-03-15:
I just pushed a change with https://github.com/mkuf/prind/commit/48287eb68bf150151dde90dfb02624df7fe6f0f5, which uses the `latest` tag instead of `development`.

This should fix the issue for you for now, but in the long run, when the development branch gets merged back into main, I suspect the arm/v6 and arm/v7 images to not be built anymore.  
If you need support for a specific architecture, you'll have to raise an issue at the moonraker-telegram-bot repository for them to add support back in.

nlef, 2024-03-15:
@mthqwork 
Currently, 32-bit systems are becoming a thing of the past and some of the bot’s dependencies are no longer supported by 32-bit distributions (it is necessary to at least rebuild many of their dependencies). Therefore, I removed the assembly of images for Armv6 and Armv7.
Explain why and in what situations builds for 32 bits are needed, especially docker images?

mkuf, 2024-03-15:
@nlef Older raspberry pi models, as well as the Zero variants use arm/v6 as far as I'm aware. This is also the reason why all images built by print support this architecture.  

As the hardware requirements to run Klipper and accompanying services is relatively low, it's common practice to repurpose lower powered and older hardware to run the applications. 

I don't know the specific download numbers for arm/v6 and arm/v7 images I publish for prind, but I can generally see recent pulls for these platforms.  
This indicates at least some usage. 

[image]

nlef, 2024-03-15:
@mkuf For me, it looks strange to run the klipper ecosystem on old weak hardware in Docker ( orangepi zero3 is realy cheap...)
Also, the telegram bot is unfortunately not lightweight and is quite demanding on resources.
In any case, it’s better to create a separate ticket in my project and I’ll think about how I can maintain compatibility with 32 bit versions

mthqwork, 2024-03-15:
Well, it's running on a RPi Zero 2W. I guess I have grabbed the 32 bit OS for it. I'll switch to 64 bit then. Not a powerhorse but it does the job very well. Even resonance measurement works. (zero1 can't do that, it's just too slow)

mkuf, 2024-03-15:
Closing this now, as this discussion is better suited for moonraker-telegram-bot's issue tracker.  
@mthqwork please open a issue there if you need further support for your desired platform. 

-Markus

closed 2024-03-15 by mkuf