mkuf/prind #138: Missing GIT for Update Manager

MrFlou, 2024-03-14:
Had to install git though "docker exec -it -u 0 prind-moonraker-1 bash" then "apt install git"

Changes to moonraker.conf
```
[update_manager]
enable_system_updates: False
```

Without this and the small change to moonraker.conf Update manager could not run. 
Running on Raspberry pi zero 2 w 64bit OS

mkuf, 2024-03-14:
Hi there,

Using the update_manager goes against the core principle of running applications in containers. The container images contain a specific version of an application and all dependencies that are required to run it.  
If you want to update the application and/or dependencies, a newer version of the container image should be used, instead of pulling updates into the existing container.  
Any changes done in the container are lost if the stack is stopped and containers are removed. 

Updating the stack is described in the readme at: https://github.com/mkuf/prind?tab=readme-ov-file#updating

Additionally, the `.git` directory that is used by update_manager to update the application is not part of the container image, so you'll start in an undefined state that gets lost with the removal of the container, bringing inconsistencies and making your setup not reproducable.
Finally, update_manager would only be able to update moonraker itself, as it has no access to the application directories that are located in the other containers (e.g. klipper). 

-Markus

MrFlou, 2024-03-14:
Ah makes sense! Though if one wants to add KAMP, how would that look?

mkuf, 2024-03-15:
Just as you would normally do.  
As stated in the readme, all config is located in the `config` directory. https://github.com/mkuf/prind?tab=readme-ov-file#configuring-klippermoonraker

You can add any additional files and directories and include them in `printer.cfg`

wrobelda, 2024-04-08:
This is overly limiting, to be honest: once can have KAMP checked out into their config file and have moonrake manage its updates with respect to container's ephemeral status.

mkuf, 2024-04-08:
Thanks for your Input.  

Thinking about it, having update manager take care of external resources besides the main components (Klipper, Moonraker, etc.) shouldn't be an issue. 

I'll prepare a PR to add git to the runtime Image of Moonraker. 

-Markus

mkuf, 2024-04-08:
Git has been added to the Dockerfile.  
The action is currently re-building the most recent moonraker image https://github.com/mkuf/prind/actions/runs/8606230353/job/23584206457

PR #144 (2024-04-08): moonraker: add git to runtime image
Adds gut to the runtime Image of Moonraker to allow update manager to be used. 

Fixes #138
commit 670659a1 (2024-04-08): moonraker: add git to runtime image (#144)

fixes #138