mkuf/prind #143: Klipper container missing dependency?

nelsongraca, 2024-04-03:
Last few versions have been failing after updating the container with this error:
Installing `setuptools` in the venv fixes this, not sure if this is an upstream problem or in the container.

    Unhandled exception during connect
    Traceback (most recent call last):
      File "/opt/klipper/klippy/klippy.py", line 176, in _connect
        self.send_event("klippy:mcu_identify")
      File "/opt/klipper/klippy/klippy.py", line 263, in send_event
        return [cb(*params) for cb in self.event_handlers.get(event, [])]
                ^^^^^^^^^^^
      File "/opt/klipper/klippy/mcu.py", line 763, in _mcu_identify
        self._serial.connect_canbus(self._serialport, nodeid,
      File "/opt/klipper/klippy/serialhdl.py", line 112, in connect_canbus
        import can # XXX
        ^^^^^^^^^^
      File "/opt/venv/lib/python3.12/site-packages/can/__init__.py", line 31, in <module>
        from .io import Logger, Printer, LogReader, MessageSync
      File "/opt/venv/lib/python3.12/site-packages/can/io/__init__.py", line 11, in <module>
        from .logger import Logger
      File "/opt/venv/lib/python3.12/site-packages/can/io/logger.py", line 13, in <module>
        from .asc import ASCWriter
      File "/opt/venv/lib/python3.12/site-packages/can/io/asc.py", line 19, in <module>
        from ..util import channel2int
      File "/opt/venv/lib/python3.12/site-packages/can/util.py", line 23, in <module>
        from can.interfaces import VALID_INTERFACES
      File "/opt/venv/lib/python3.12/site-packages/can/interfaces/__init__.py", line 8, in <module>
        from pkg_resources import iter_entry_points
    ModuleNotFoundError: No module named 'pkg_resources'

nelsongraca, 2024-04-03:
I can spare  some time to submit the PR to Klipper if you don't mind.

mkuf, 2024-04-04:
Sure, go ahead 🙂

nelsongraca, 2024-04-05:
Done in https://github.com/Klipper3d/klipper/pull/6550

nelsongraca, 2024-04-09:
New PR on Klipper https://github.com/Klipper3d/klipper/pull/6557

nelsongraca, 2024-04-10:
@mkuf considering how long getting the fix on Klipper is taking, can we add the latest version of python-can as an extra dependency on the container build process and remove when this is fixed upstream?

mkuf, 2024-04-10:
I don't think its a good idea to upgrade a package that klipper depends on here instead of upstream.  

As it was stated in your first PR, a viable workaround is to add setuptools without any version pinning to the venv.  
I'll prepare a PR to add this to the klipper image.

mkuf, 2024-04-10:
PR has been merged. 
setuptools will be included with the next images built by the daily github action run.  

I'll leave this issue open for now to observe process in your upstream PRs.

mkuf, 2024-05-05:
In the light of this issue and also https://github.com/mkuf/prind/issues/150, I decided that I will revert the klipper base image back to python 3.11, as it does seem that python 3.12 is a bit too bleeding edge for klipper and I favor reliability over up-to-dateness. For now, at least.

-Markus

closed 2024-05-05 by mkuf