mkuf/prind #148: Moonraker crash after update

tyler43636, 2024-04-11:
I performed an update a few hours ago and now moonraker appears to be crashing every few seconds. A consistent experience across browsers is that the Mainsail or Fluidd interface will load, allow for full interaction with klipper (set temps, home axis, etc.) and then Mainsail or Fluidd will loose connection to moonraker and reload within a couple seconds. 

Below is my moonraker log. Any ideas?

```
-------------------- Log Start | Thu Apr 11 10:01:16 2024 --------------------
platform: Linux-5.15.111-1-pve-x86_64-with-glibc2.36
data_path: /opt/printer_data
is_default_data_path: False
config_file: /opt/printer_data/config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: 814d0179c6fd4eb9ad7248b6c6b79fa6
unix_socket_path: /opt/printer_data/comms/moonraker.sock
software_version: ?
log_file: /opt/printer_data/logs/moonraker.log
python_version: 3.11.8 (main, Mar 12 2024, 11:52:02) [GCC 12.2.0]
launch_args: /opt/venv/bin/python moonraker/moonraker/moonraker.py -d /opt/printer_data
msgspec_enabled: False
uvloop_enabled: False
2024-04-11 10:01:16,997 [confighelper.py:read_file()] - Configuration File '/opt/printer_data/config/moonraker.conf' parsed, total size: 439 B
2024-04-11 10:01:16,997 [server.py:add_log_rollover_item()] - 
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /opt/printer_data/run/klipper.sock

[machine]
provider = systemd_cli
validate_service = False

[authorization]
trusted_clients = 
	0.0.0.0/0
cors_domains = 
	*

[octoprint_compat]

[history]

#################################################################
All Configuration Files:
/opt/printer_data/config/moonraker.conf
#################################################################
2024-04-11 10:01:17,062 [server.py:load_component()] - Component (secrets) loaded
2024-04-11 10:01:17,074 [server.py:load_component()] - Component (template) loaded
2024-04-11 10:01:17,077 [server.py:load_component()] - Component (klippy_connection) loaded
2024-04-11 10:01:17,717 [server.py:load_component()] - Component (application) loaded
2024-04-11 10:01:17,749 [server.py:load_component()] - Component (websockets) loaded
2024-04-11 10:01:17,890 [server.py:load_component()] - Component (dbus_manager) loaded
2024-04-11 10:01:17,909 [server.py:add_log_rollover_item()] - 
Database Versions: pylmdb = 1.4.1, lmdb = (0, 9, 29)
Unsafe Shutdown Count: 0
2024-04-11 10:01:17,915 [server.py:load_component()] - Component (database) loaded
2024-04-11 10:01:17,979 [file_manager.py:__init__()] - Using File System Observer: inotify
2024-04-11 10:01:17,985 [server.py:load_component()] - Component (file_manager) loaded
2024-04-11 10:01:18,005 [authorization.py:__init__()] - Authorization Configuration Loaded
Trusted Clients:
0.0.0.0/0
CORS Domains:
.*
2024-04-11 10:01:18,006 [server.py:load_component()] - Component (authorization) loaded
2024-04-11 10:01:18,012 [server.py:load_component()] - Component (klippy_apis) loaded
[... 113 lines cut ...]
  File "/opt/moonraker/moonraker/components/shell_command.py", line 277, in run_with_response
    raise ShellCommandError(
moonraker.components.shell_command.ShellCommandError: Error running shell command: 'systemctl status 1'
2024-04-11 10:01:21,176 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
2024-04-11 10:01:21,179 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
2024-04-11 10:01:21,428 [job_state.py:_handle_started()] - Job state initialized: standby
2024-04-11 10:01:21,678 [base_events.py:default_exception_handler()] - Future exception was never retrieved
future: <Future finished exception=EOFError()>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/aio/message_bus.py", line 158, in on_hello
    raise err
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/aio/message_bus.py", line 365, in _message_reader
    if self._unmarshaller.unmarshall():
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 304, in unmarshall
    self._unmarshall()
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 243, in _unmarshall
    self.read(16, prefetch=True)
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 91, in read
    raise EOFError()
EOFError
2024-04-11 10:01:21,686 [klippy_connection.py:_check_ready()] - Klippy ready
2024-04-11 10:01:22,181 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'extruder']

```

mkuf, 2024-04-11:
Hey there,

please run `scripts/get-info.sh` and share the generated archive.

-Markus

tyler43636, 2024-04-17:
Thanks Markus. When I return and ran another docker pull everything was working again. I’ll close this issue.

closed 2024-04-17 by tyler43636

nrgbistro, 2024-11-10:
I'm also running into this same issue. My setup worked great for a while. I hadn't used my printer for a month and tried access prind tonight just to see this error in the logs moonraker. I haven't changed anything about my setup. The printer is on with the usb connected, and the `/dev/ender3` device is definitely mapped correctly. I tried completely recreated the containers, as well as restoring my prind data folder from a backup but can't get past this. I'm running on Unraid 6.12.13 using the latest version of prind.

```
Application Info:
platform: Linux-6.1.106-Unraid-x86_64-with-glibc2.36
data_path: /opt/printer_data
is_default_data_path: False
config_file: /opt/printer_data/config/moonraker.conf
backup_config: /opt/printer_data/config/.moonraker.conf.bkp
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: 33bd616842f44d48a59e274e208a705f
unix_socket_path: /opt/printer_data/comms/moonraker.sock
software_version: v0.9.3-1-g4e00a07-prind
log_file: /opt/printer_data/logs/moonraker.log
python_version: 3.12.7 (main, Oct 19 2024, 01:09:09) [GCC 12.2.0]
launch_args: /opt/venv/bin/python moonraker/moonraker/moonraker.py -d /opt/printer_data
msgspec_enabled: False
uvloop_enabled: False
[confighelper.py:read_file()] - Configuration File '/opt/printer_data/config/moonraker.conf' parsed, total size: 490 B
[server.py:add_log_rollover_item()] - 
#################### Moonraker Configuration ####################
[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /opt/printer_data/run/klipper.sock
[machine]
provider = systemd_cli
validate_service = False
[authorization]
trusted_clients = 
	0.0.0.0/0
cors_domains = 
	*
[octoprint_compat]
[history]
[file_manager]
enable_object_processing = True
#################################################################
All Configuration Files:
/opt/printer_data/config/moonraker.conf
#################################################################
[server.py:load_component()] - Component (secrets) loaded
[server.py:load_component()] - Component (template) loaded
[server.py:load_component()] - Component (klippy_connection) loaded
[server.py:load_component()] - Component (application) loaded
[server.py:load_component()] - Component (websockets) loaded
[server.py:add_log_rollover_item()] - Loading Sqlite database provider. Sqlite Version: 3.40.1
[server.py:add_log_rollover_item()] - Database: Stored Instance ID does not match current Instance ID.
Current UUID: 33bd616842f44d48a59e274e208a705f
Stored UUID: 12769413d1db4483a29555c35910db43
[server.py:add_log_rollover_item()] - Unsafe Shutdown Count: 6
[server.py:load_component()] - Component (database) loaded
[server.py:load_component()] - Component (dbus_manager) loaded
[file_manager.py:__init__()] - Using File System Observer: inotify
[server.py:load_component()] - Component (file_manager) loaded
[database.py:register_table()] - Found registered table authorized_users
[authorization.py:__init__()] - Authorization Configuration Loaded
Trusted Clients:
0.0.0.0/0
CORS Domains:
.*
[server.py:load_component()] - Component (authorization) loaded
[server.py:load_component()] - Component (klippy_apis) loaded
***distribution***
  name: Debian GNU/Linux 12 (bookworm)
[... 80 lines cut ...]
  File "/opt/moonraker/moonraker/components/shell_command.py", line 277, in run_with_response
    raise ShellCommandError(
moonraker.components.shell_command.ShellCommandError: Error running shell command: 'systemctl status 5470'
[server.py:add_log_rollover_item()] - Klipper Version: v0.12.0-349-ga34034494-prind
[shell_command.py:_check_proc_success()] - Command (systemctl status 1) exited with return code 1
[machine.py:extract_service_info()] - Error extracting service info
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/components/machine.py", line 1097, in extract_service_info
    resp: str = await self.shell_cmd.exec_cmd(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/components/shell_command.py", line 277, in run_with_response
    raise ShellCommandError(
moonraker.components.shell_command.ShellCommandError: Error running shell command: 'systemctl status 1'
[klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
[klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
[klippy_connection.py:_check_ready()] - 
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
[server.py:_handle_term_signal()] - Exiting with signal SIGTERM
[klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
[server.py:main()] - Server Shutdown
```

mkuf, 2024-11-11:
@nrgbistro the support file does not contain all necessary information, as you're using a custom compose file.  
Judging from the moonraker log you provided, I think the following lines are relevant: 

```
[...]
[server.py:_initialize_component()] - Performing Component Post Init: [dbus_manager]
[dbus_manager.py:component_init()] - Unable to Connect to D-Bus
[...]
[shell_command.py:_check_proc_success()] - Command (systemctl status 1) exited with return code 1
[...]
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/components/machine.py", line 1097, in extract_service_info
    resp: str = await self.shell_cmd.exec_cmd(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/components/shell_command.py", line 277, in run_with_response
    raise ShellCommandError(
moonraker.components.shell_command.ShellCommandError: Error running shell command: 'systemctl status 1
```

I'm unfamiliar with unraid but it seems it is not using systemd, which is causing issues.  
Try setting the machine provider to `none` in your moonraker.conf like so:

```
[machine]
provider: none
```

nrgbistro, 2024-11-12:
Thanks for your reply! I'm starting to think that wasn't actually an issue, and I just thought it was due to other things with my setup breaking. After figuring out how to properly mount my printer to `/dev/ender3`, I'm facing some permission issues that I'm struggling to solve. Klipper is clearly able to see the device, which I've confirmed by running `ls -al` in the container, but for some reason it doesn't have permission to use it. Do you have any ideas on how I might be able to fix this?

`mcu 'mcu': Unable to open serial port: [Errno 13] could not open port /dev/ender3: [Errno 13] Permission denied: '/dev/ender3'`

On Unraid, the `/dev/ender3` device has 777 permissions and is owned by `root root`. Inside the container it only seems to have 700 permissions set, but I can't figure out why this is or how to change it.

mkuf, 2024-11-12:
@nrgbistro 

This is probably because you're adding `/dev/ender3` as a volume to your klipper container.  
Try mounting `/dev` instead, as the usage section for that image suggests https://github.com/mkuf/prind/tree/main/docker/klipper#usage

https://github.com/mkuf/prind/blob/57a1e1dc985788b8c032f9b7e20bef4c8464c323/docker-compose.yaml#L20-L25

Also, there's an section about Device permissions in the main readme that might help https://github.com/mkuf/prind#device-permissions

nrgbistro, 2024-11-13:
Great, thank you again for your patience helping me out with this. I've had so many problems with Unraid (both related to prind and otherwise), I think for now I'm going to use Kiauh on a raspberry pi. This is an awesome project though and I definitely plan to come back to it some day when I have a better server and more printers, so thanks for creating, maintaining, and supporting it!!