mkuf/prind #150: Fatal Python error: PyGILState_Release: auto-releasing thread-state, but no thread-state for this thread

ginkel, 2024-04-18:
Hi there,

I am using your `mkuf/klipper` Docker image for my Docker-based Klipper setup on a Raspi 4. Starting with a recent upgrade the container started crashing when my 3D printer becomes idle and powers off. The following error is logged:

```
Fatal Python error: PyGILState_Release: auto-releasing thread-state, but no thread-state for this thread
Python runtime state: initialized

Thread 0x0000ffff9ef5f180 (most recent call first):
  <no Python frame>

Thread 0x0000ffff9e74f180 (most recent call first):
  File "/opt/klipper/klippy/serialhdl.py", line 39 in _bg_thread
  File "/usr/local/lib/python3.12/threading.py", line 1010 in run
  File "/usr/local/lib/python3.12/threading.py", line 1073 in _bootstrap_inner
  File "/usr/local/lib/python3.12/threading.py", line 1030 in _bootstrap

Thread 0x0000ffffa065f180 (most recent call first):
  File "/usr/local/lib/python3.12/threading.py", line 355 in wait
  File "/usr/local/lib/python3.12/queue.py", line 171 in get
  File "/opt/klipper/klippy/queuelogger.py", line 34 in _bg_thread
  File "/usr/local/lib/python3.12/threading.py", line 1010 in run
  File "/usr/local/lib/python3.12/threading.py", line 1073 in _bootstrap_inner
  File "/usr/local/lib/python3.12/threading.py", line 1030 in _bootstrap

Thread 0x0000ffffa1b89020 (most recent call first):
  File "/opt/klipper/klippy/reactor.py", line 342 in _dispatch_loop

Extension modules: greenlet._greenlet, _cffi_backend, markupsafe._speedups (total: 3)
```

Has anyone else seen this? Could possibly be related to the Python 3.12 upgrade, but that's just a wild guess...

mkuf, 2024-04-20:
This might be related to the greenlet upgrade that was introduced in https://github.com/Klipper3d/klipper/pull/6525 to allow for the klippy-requirements to be compatible with python 3.12. 

Which tag of the klipper image are you using?
If `latest`, you can find the klipper version with the following command
```
docker inspect --format '{{ index .Config.Labels "org.prind.image.version"}}' mkuf/klipper:latest
```

Could you describe in more detail under which conditions the error is occuring?
Is the printer shutting down by itself when no print job is running?

If possible, please share a full klippy logfile.  

-Markus

ginkel, 2024-04-21:
Hi Markus,

thanks for your reply! Regarding your questions:

I am using `mkuf/klipper:latest`.

```
$ docker inspect --format '{{ index .Config.Labels "org.prind.image.version"}}' mkuf/klipper:latest
v0.12.0-159-g2425a746
```

I have configured an idle timeout to power off my printer after idling for ten minutes:

```
[idle_timeout]
gcode =
    TURN_OFF_HEATERS
    M84
    POWER_OFF_PRINTER
timeout = 600

[gcode_macro POWER_OFF_PRINTER]
gcode =
  {action_call_remote_method(
    "set_device_power", device="printer", state="off"
  )}
```

This turns off a Zigbee plug controlled via Home Assistant effectively turning off the 3D printer (but not the Raspi running Klipper).

Most of the time (but not always) after the printer has turned off due to being idle the klipper container terminates with the Python error quoted above. For completeness, I am attaching my `klippy.log` along with the `docker logs` output of the container (although the log isn't very insightful).

The issue happened after the most recent print finished.

[docker-logs.txt](https://github.com/mkuf/prind/files/15053092/docker-logs.txt)
[klippy.log](https://github.com/mkuf/prind/files/15053093/klippy.log)

Please let me know if you need any further details!

Thanks,
Thilo

mkuf, 2024-04-23:
I have really no idea whats causing this.  

Does the crash/restart stop if the printer is turned back on?  
Does this have any negative impact besides the container restarting?

Could you give `mkuf/klipper:issue-150_v0.12.0-159-g2425a746` a spin and see if the issue persists?
I built the same version you're currently using but with python 3.11 as base image.

ginkel, 2024-04-28:
I deployed your image on Apr 24 and did a couple of prints / power cycles sine then. The problem did not manifest again so far.

mkuf, 2024-05-05:
Thanks for giving it a try. 

In the light of this issue and also https://github.com/mkuf/prind/issues/143, I decided that I will revert the klipper base image back to python 3.11, as it does seem that python 3.12 is a bit too bleeding edge for klipper and I favor reliability over up-to-dateness. For now, at least. 

To make klipper work on 3.12, the upstream devs would greatly benefit from a description of the issues you encountered, so I'd klindly ask you to reach out to them via their [contact points](https://github.com/Klipper3d/klipper/blob/master/docs/Contact.md) and provide them this valuable information. 

-Markus