mkuf/prind #165: Not compatible with Portainer

veeroohre, 2024-07-10:
I use Portainer for managing my docker container.

Unfortunately the docker compose of prind uses relativ pathes ( ./config) and mounts single files.
This is unfortunately not compatible with the volume approach of Portainer.

Would it be possible to develop a composer file with only volumes? I tried it myself but failed.

mkuf, 2024-07-10:
The `custom` directory contains a compose file that is compatible with portainer at https://github.com/mkuf/prind/blob/main/custom/docker-compose.custom.portainer.yaml

The comment at the top of the file describes the required steps to get the services up and running. 

-Markus

veeroohre, 2024-07-10:
Didn't see that. Sorry!

Danke Markus!

veeroohre, 2024-07-10:
In the klipper section in this file there's still a relative mount:
      - ./config:/opt/printer_data/config

Would it possible to move that to a volume or "/data/prind/config"?

mkuf, 2024-07-10:
Good catch, that path should also point to `/data/prind/config` instead of `./config`.  
I fixed the path with https://github.com/mkuf/prind/commit/0362c28b41ed26a106fe17aa56d600029d60388c

veeroohre, 2024-07-10:
Mainsail and Octoprint are missing in this file or is it on purpose?

mkuf, 2024-07-10:
Yes, this is by design.  

As the main compose file is incompatible w/ portainer, die custom file defines just a basic stack with static service definitions to provide a starting point for portainer users. Other services may be added to the portainer file by referencing the main compose file.

closed 2024-07-12 by mkuf