mkuf/prind #177: Add healthcheck to the mkuf/ustreamer image

PavelD, 2024-11-05:
Create standalone healthcheck script returning true / false based on the service availability.

Optionally provide guide what and how to check to get the same result using installed tools. Usually docker images contains curl, busybox (wget) or netcat so it's possible to build healthcheck by checking the HTTP endpoint. This image has no such tool installed.

mkuf, 2024-11-05:
Could you elaborate on your usecase for this?   
Is the health check meant to help with external monitoring, or is it intended to handle specific cases where the service needs frequent restarts?

A simple file descriptor in bash may be used to check if a port is available like so:
```bash
#!/bin/sh

HOST="localhost"
PORT="8080"

## Use file descriptor to open a socket to the host an port
if (exec 3<>/dev/tcp/$HOST/$PORT) 2>/dev/null; then
  ## Close file descriptor
  exec 3<&-
  exec 3>&-

  ## success
  exit 0
else
  ## failure
  exit 1
fi
```

PavelD, 2024-11-05:
Hi, sorry. I did mistake when i was trying that. I had FQDN + local port instead of 443. My bad.

closed 2024-11-05 by PavelD

PavelD, 2024-11-05:
Here is inliner for the docker-compose file.
```
healthcheck:
      test: ["CMD", "/bin/bash", "-c", "if (exec 3<>/dev/tcp/localhost/8080) 2>/dev/null; then exec 3<&-; exec 3>&-; exit 0; else exit 1; fi"]
      interval: 30s
      timeout: 10s
      retries: 3
```

mkuf, 2024-11-12:
Advanced Health checks have been implemented with https://github.com/mkuf/prind/commit/9ef0e9df1abee23fd553b2018f1ad2f8995e3529 for klipper, moonraker and ustreamer. 

You may have to re-download the latest images from dockerhub to get the changes.