mkuf/prind #180: Connection to Moonraker fails

gercamjr, 2024-11-21:
I am using the mainsail profile with my Creality Ender 3 v2 Neo printer. I had no issues with Mainsail connecting to Moonraker and bringing up the dashboard. Even printed a few items. Today I tried updating the printer.conf for my printer, and clicked the Save & Restart button in the dashboard. I got an error message saying "**_Can not update MCU 'mcu' config as it is shutdown_**" So I clicked on the Firmware_Restart button and got the same message. Then I clicked on the Restart button and the page displays that the connection to Moonraker fails. I tried pulling the latest images, restarting various times, undoing the small edit to the printer.conf, restarting all containers, and still the same thing. I checked the Klippy logs and it reports it detects the printer and can connect, etc.

[image]

[prind-info-21112024-173749.tar.gz](https://github.com/user-attachments/files/17849932/prind-info-21112024-173749.tar.gz)

mkuf, 2024-11-21:
Hi,

unfortunately, the Archive is missing the application logs.  
Please share your moonraker.log and klippy.log.  

-Markus

gercamjr, 2024-11-21:
Thanks for replying. Somehow it all resolved itself when I unplugged the
USB cable to the printer, turned it off, plugged it back in and turned it
back on. Moonraker had no issue afterwards. Not sure why though.

Thanks,

mkuf, 2024-11-21:
Great, I'll close this issue then.

closed 2024-11-21 by mkuf