mkuf/prind #182: Can this be configured to allow access to the Moonraker UI?

Nite16, 2024-12-05:
My stack is up and running, but I can't seem to view the Moonraker UI directly, can this be setup with Traefik?

mkuf, 2024-12-06:
Hello, 

The simplest way would be to open the moonraker port on your host machine.  
To do so, add the following to your `docker-compose.override.yaml`.  
The Moonraker Web UI will be available at `http://<yourhost>:7125`
```yaml
services:
  moonraker:
    ports:
      - 7125:7125
```

If you explicitly want to publish the moonraker UI via traefik, add the following to your `docker-compose.override.yaml`.  
The Moonraker Web UI will be available at `http://<yourhost>/moonraker`.  
You may have to clear your browser cache after the rule is active, otherwise you may get forwarded to Mainsail/Fluidd. 

```yaml
services:
  moonraker:
    labels:
      traefik.http.routers.moonraker-middleware.rule: PathPrefix(`/moonraker`)
      traefik.http.routers.moonraker-middleware.entrypoints: web
      traefik.http.routers.moonraker-middleware.middlewares: moonraker-strip-prefix
      traefik.http.routers.moonraker-middleware.service: moonraker
      traefik.http.middlewares.moonraker-strip-prefix.stripPrefix.prefixes: /moonraker
```

Nite16, 2024-12-06:
Thanks! I ended up doing option 1, which is working well.

closed 2024-12-07 by mkuf