mkuf/prind #194: Install addons/plugins

mircsicz, 2025-01-20:
Hi Markus, erstmal 1000 Dank für Dein tolles Projekt! <3

Thanks to prind I could klipper my Kobra 2 Neo in a breeze...

But after I've started with Kiauh to build my firmware and for first test and also install the OctoApp-Plugin and a few others. But as a fellow DevOps I had to switch to Docker to run my environment! But now I can't find a single hint on how to install stuff like:

- [OctoApp-Plugin](https://github.com/crysxd/OctoApp-Plugin/wiki/Installation-on-Klipper)
- [Crowsnest](https://github.com/mainsail-crew/crowsnest)
- [gcode_shell_command](https://github.com/th33xitus/kiauh/blob/master/docs/gcode_shell_command.md)
to only name a few...

I would expect them to be installed in specific dir but I can't see where... Except for config there's not a single Volume/Dir that would make sense to me.

So I guess one could come up with a new Volume to be mounted into the klipper-image... Hope you have a hint for me ;-)

mkuf, 2025-01-20:
Hi there, 

that depends on how that "plugin" interacts with moonraker/klipper. 

If files need to be placed in a specific directory of one of the services, like the gcode_shell_command, its best do add a build section to your `docker-compose.override.yaml` and write a short dockerfile that references the prind image. 

```yaml
services:
  klipper:
    image: klipper:gsc
    build:
      dockerfile_inline: |
        FROM alpine/curl AS download
        RUN curl https://raw.githubusercontent.com/dw-0/kiauh/refs/heads/master/resources/gcode_shell_command.py -o /tmp/gcode_shell_command.py

        FROM mkuf/klipper:latest
        COPY --chown=klipper:klipper --from=download /tmp/gcode_shell_command.py /opt/klipper/klippy/extras/
```

Other "plugins" like crowsnest are separate processes that would run alongside klipper/moonraker. 
The cleanest solution for this case is to contribute a Dockerfile / Image build pipeline to the upstream project.  
We can then add a new service to the compose file referencing that image.

I did something like that for mobileraker_companion in https://github.com/Clon1998/mobileraker_companion/pull/29  
If the upstream project has no interest in maintaining a docker image, like it was the case with klipper an moonraker, we can think about adding it to prind itself and publish these images for everyone to use. 

Aside from that, I'm not really familiar with crowsnest and its usecases.  
Prind has a way to set up multiple webcams, as described here https://github.com/mkuf/prind?tab=readme-ov-file#multiple-webcams
I'd be happy to hear about your specific usecase and if it can be covered by the webcam templating. 

I also had a quick glance at the OctoApp companion but couldn't identify what the installer is doing without having to test it myself. If you have insights on that, I may be able to also provide an example on how to implement it.

mircsicz, 2025-01-20:
I'll have to get the Kids from school now ;-)

I'm in EST and as I guess your in CET it might be (at least for you) tonight until I report back...

mircsicz, 2025-01-22:
Sorry for the delay, family first... Won't be better next days as 2 out of 3 kids will have their birthdays in the next 5 days...

Here's a glance at the [install.sh](https://github.com/crysxd/OctoApp-Plugin/blob/release/install.sh)
```
#
# OctoApp for Klipper!
#
# Use this script to install the plugin on a normal device or a Creality device, or to install the companion!
# For a companion install, use the -companion argument.
#
# Simply run ./install.sh from the git repo root directory to get started!
```

mkuf, 2025-01-23:
Hey again, 

so i had a look at the octoapp installer and companion for moonraker and it would probably be possible to package it into its own container as it communicates with moonraker via its network port but it would require upstream code changes to work.

Unfortunately the address and port are hardcoded to localhost in the client, as it assumes it is running on the same host as moonraker itself.  
https://github.com/crysxd/OctoApp-Plugin/blob/54f726701b032e15184ca80a0c42aeab5ad93441/moonraker_octoapp/moonrakerclient.py#L83   

When running in a different container, it would be required to point the companion to the moonraker container instead of localhost. 

I'd suggest giving the [mobileraker](https://github.com/mkuf/prind?tab=readme-ov-file#mobileraker_companion) or [moonraker-obico](https://github.com/mkuf/prind?tab=readme-ov-file#moonraker-obico) profile a try for now.

mircsicz, 2025-01-29:
I've now run mobileraker for a few days/prints and am quite happy so far! Thx a ton for the hint ;-)

I'll activate mobileraker-companion profile in the coming days and I love it's camera implementation! I will give the custom settings for [timelaps](https://github.com/mkuf/prind/issues/195) a go as well...

Crowsnest expectation was fullfilled by the default camera implementation from mainsail... As I'm running go2rtc on a different Pi5 there was no need to look into your offering for that.

Thanks a ton once again ;-)

closed 2025-01-29 by mircsicz