mkuf/prind #195: Moonraker timelapse add-on

Pneumanifest, 2025-01-28:
Would it be possible to add a script for installing Moonraker timelapse? Is that even possible with a docker setup?

mkuf, 2025-01-28:
Hi there,

This has been discussed in https://github.com/mkuf/prind/issues/46.  
There is a override file in the [custom](https://github.com/mkuf/prind/blob/main/custom/docker-compose.custom.moonraker-timelapse.override.yaml) directory of this repo that contains all info required to get this up and running. 

-Markus

mircsicz, 2025-01-29:
May I?

I've been following your description in [docker-compose.custom.moonraker-timelapse.override.yaml](https://github.com/mkuf/prind/blob/main/custom/docker-compose.custom.moonraker-timelapse.override.yaml) and added this to docker-compose.override.yaml:
```
services:
  moonraker:
    image: moonraker:ffmpeg
    build:
      dockerfile_inline: |
        FROM alpine/git as pull
        RUN git clone https://github.com/mainsail-crew/moonraker-timelapse /opt/moonraker-timelapse
        FROM mkuf/moonraker:latest
        USER root
        RUN apt update \
        && apt install -y ffmpeg wget \
        && apt clean
        USER moonraker
        COPY --chown=moonraker:moonraker --from=pull /opt/moonraker-timelapse /opt/moonraker-timelapse
        RUN ln -s /opt/moonraker-timelapse/component/timelapse.py /opt/moonraker/moonraker/components/timelapse.py
    volumes:
      - /opt/timelapse:/opt/timelapse
      - /dev/null:/opt/klipper/config/null
      - /dev/null:/opt/klipper/docs/null
      - /run/dbus:/run/dbus
      - /run/systemd:/run/systemd
      - run:/opt/printer_data/run
      - gcode:/opt/printer_data/gcodes
      - log:/opt/printer_data/logs
      - moonraker-db:/opt/printer_data/database
      - ./config:/opt/printer_data/config
```
and yes I've skipped ustreamer / webcam as I run a go2rtc instance on another machine...

I've also added a timelapse.cfg to my ./config dir and the line
```
[include timelapse.cfg]
``` 
as well as the [file](https://github.com/mainsail-crew/moonraker-timelapse/blob/main/klipper_macro/timelapse.cfg).

Also my moonraker.conf now contains:
```
[timelapse]
output_path: /opt/timelapse/
##
[webcam]
snapshot_url: https://subdom.mydom.net/go2rtc/api/frame.jpeg?src=kobra2neo-rstp2
```

But with the [timelapse] part active moonraker won't start:
```
2025-01-29 15:54:48,795 [server.py:try_pip_recovery()] - Module 'moonraker.components.timelapse' not found. Attempting Pip Update...
```

I've also already removed the klipper image... 

To me it seems that the part from override.yaml is ignored. :-(

mkuf, 2025-01-29:
@mircsicz is that your full override file? If not, please upload the file here. 

Did you see any build action when starting the stack?  
If not, you may need to run `docker compose build` and append your profiles for it to work.

mircsicz, 2025-01-29:
@mkuf yeap thats all there's to it ;-)

I couldn't se any build when starting with:
`docker` compose --profile mainsail,mobileraker_companion up -d

And a 'docker compose --progress plain build' also left no output at all

mkuf, 2025-01-29:
@mircsicz The way you specify profiles in your `up` command is invalid.  
You can see that on the number of containers running, mainsail and the mobileraker_companion are missing.  

Moonraker is also missing, as it is tied to a specific set of profiles, because it would not be needed if one would choose octoprint as frontend. 

https://github.com/mkuf/prind/blob/39e391eedc3f5150c3f130df0015317163b71b06/docker-compose.yaml#L52-L57

The profile argument needs to be repeated for every profile, so the correct syntax for you would be as outlined in the readme
https://github.com/mkuf/prind/blob/39e391eedc3f5150c3f130df0015317163b71b06/README.md#L171

mircsicz, 2025-01-29:
Just Found it in your docs, SORRY once again RTFM 🤨

Pneumanifest, 2025-01-30:
Yeah I think I'll be able to get it going. Thank you for the follow up.

closed 2025-01-30 by Pneumanifest