mkuf/prind #196: Expose Moonraker Port for use with GetHomepage

mircsicz, 2025-01-30:
I've always had Octoprint integrated into my 'Homepage'

Now that I klipperized my Kobra2Neo and migrated to Mainsail I wanted to change to the Moonraker [widget](https://gethomepage.dev/widgets/services/moonraker/) 

```
    - Klipper:
          href: http://10.10.xxx.19
          widget:
            type: moonraker
            url: http://10.10.xxx.19:7125
            fields: ["printer_state","print_status","print_progress","layers"]
```
so I've tried to adjust in docker-compose.yaml:

```
  ## WebApi
  ##
  moonraker:
    image: mkuf/moonraker:latest
    restart: unless-stopped
    pid: host
    logging:
      driver: none
    #ports:
    #  - "7125:7125"
```
But with the 'ports:' statement being uncommented I can't reach the port...

I don't have a lot of experience using labels, so I hope you can give me hint what I need to adjust instead of 'ports'.

mkuf, 2025-01-30:
Moonraker is exposed via the traefik proxy at port 80.  
Just remove the 7125 from your widget config and you're good to go.

mircsicz, 2025-01-30:
I was afraid you say so 🤣

[image]

Seems I'll have to open a ticket over at [GetHomepage](https://github.com/gethomepage/homepage)

closed 2025-01-31 by mircsicz