mkuf/prind #198: Limit /dev access to moonraker container

thesolomon-tech, 2025-01-31:
Hi,

In the Moonraker Docker container the /dev folder is being shared in its entirety. Is there a way to limit access to just the ports that connect to the printer?

Thanks

mircsicz, 2025-01-31:
Here's an example from octorpint, that's how I ran it in Docker before I migrated to prind:

```
---
services:
  octoprint:
    image: octoprint/octoprint
    restart: unless-stopped
    hostname: octo
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    volumes:
      - /srv/docker/octoprint/config:/octoprint
    networks:
      - reverseproxy

networks:
  reverseproxy:
    external: true
    name: caddy
```

mkuf, 2025-01-31:
This was introduced as a fix for https://github.com/mkuf/prind/issues/77 to improve user experience.

The issue here is that a device that is located at `/dev/ttyUSB0` may be available under a different device file after it has been  reconnected to the host (e.g. `/dev/ttyUSB1`).  
This is why the official klipper docs use the static `/dev/serial/by-id` symlink instead of the device directly.  
Having `/dev` mounted to the klipper container allows the usage of the symlink and therefore prevents the setup from breaking if the mcu is reconnected. 

If you're certain that your device file will never change, you may override the volumes and add a `device` directive to your `docker-compose.override.yaml` like so: 

```yaml
services:
  klipper:
    volumes:
      - ./config:/opt/printer_data/config
      - run:/opt/printer_data/run
      - gcode:/opt/printer_data/gcodes
      - log:/opt/printer_data/logs
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
```

thesolomon-tech, 2025-02-03:
Thanks for the help @mkuf @mircsicz

closed 2025-02-03 by thesolomon-tech