mkuf/prind #203: Documentation unclear on how the make alias works

sixtyfive, 2025-02-15:
```
jrs@bender:~/prind $ make menuconfig
ERROR: No such service: make
```

How is `alias make="docker-compose -f $HOME/prind/docker-compose.override.yaml run --rm make"` intended?

mkuf, 2025-02-15:
Hi, 

the alias is just there to simplify the commands one has to run and make them adaptable to the official klipper docs.  

The container is defined in `docker-compose.extra.make.yaml`, and has no ties to the main docker-compose or override file, so the command you shared can't work.  

Furthermore, that alias only works when you're in the main directory of prind and is not intended to work system wide. This is just an ad-hoc alias for the current session and should be set up each time one wants to build the mcu code. 

https://github.com/mkuf/prind/blob/f6adf10a45afa254bc357cb1d463eb478ce76f72/README.md#L49

Let me know if you have any further questions.  
-Markus

closed 2025-02-23 by mkuf

limbo99wastaken, 2025-03-21:
Hi, 

I think I'm running into the same issue. I ran 
 alias make="docker compose -f docker-compose.extra.make.yaml run --rm make" 

and then 
make menuconfig

when I run it with sudo I get
make: *** No rule to make target 'menuconfig'.  Stop.

and when I run it without sudo I get
permission denied while trying to connect to the Docker daemon socket at 

thanks 
 -a linux newbie

mkuf, 2025-03-22:
Hi @limbo99wastaken, 

this is pretty unintuitive, but aliases do not work with sudo.

The underlying issue is that your current user does not have permission to talk to the docker daemon via its socket - thats the error message you received when you ran it without sudo. 
Add your user to the docker group using the following command and replacing `$USER` with your actual username. 

```
sudo usermod -aG docker $USER
```
This is also outlined in the [official docker docs](https://docs.docker.com/engine/install/linux-postinstall/#manage-docker-as-a-non-root-user). 

-Markus