mkuf/prind #211: No moonraker connection.

Tabaqui, 2025-04-24:
Hello!

So I'm trying to get this going on my arch Linux machine.

I do 

`make menuconfig`
`make`

it is seems ok as well as firmware but

`docker compose --profile fluidd up -d`

gives me this

`✔ Network prind_default        Created
 ⠹ Container prind-webcam-1     Starting
 ✔ Container prind-traefik-1    Started
 ✔ Container prind-init-1       Started
 ✔ Container prind-fluidd-1     Started
 ✔ Container prind-klipper-1    Created
 ✔ Container prind-moonraker-1  Created`

which is weird but I'm not that familiar with docker. And trying fluidd in web gives nothing but message "No moonraker connection" with two buttons to refresh which lead the same

mkuf, 2025-04-24:
Hey,

When running on non Debian based systems, it might be necessary to set specific device permissions, see https://github.com/mkuf/prind?tab=readme-ov-file#device-permissions for reference.

Other than that, please share the output of `docker compose ps ` and the log files created by Klipper and Moonraker, which should be located in `/var/lieb/docker/volumes/prind_log/_data`

Tabaqui, 2025-04-25:
I don't have any files in `/var/lib/docker/volumes/prind_log/_data`.

Regarding `docker compose ps':
```
NAME              IMAGE                               COMMAND                  SERVICE   CREATED       STATUS       PORTS
prind-fluidd-1    ghcr.io/fluidd-core/fluidd:latest   "/docker-entrypoint.…"   fluidd    7 hours ago   Up 7 hours   80/tcp
prind-traefik-1   traefik:3.3                         "/entrypoint.sh --ac…"   traefik   7 hours ago   Up 7 hours   0.0.0.0:80->80/tcp, [::]:80->80/tcp
```
and rules(permissions) is:
`ACTION=="add",SUBSYSTEM=="tty",ATTRS{idVendor}=="1d50",ATTRS{idProduct}=="614e",GROUP="20"`
which corresponds `llsusb` output:
`Bus 001 Device 007: ID 1d50:614e OpenMoko, Inc. stm32g0b1xx`

mkuf, 2025-04-25:
Please add the following to your `docker-compose.override.yaml`, then stop and start the stack again. 

```yaml
services:
  klipper:
    logging:
      driver: json-file

  moonraker:
    logging:
      driver: json-file
```

After the stack has started, execute the following commands and share the **output** you received.

* docker compose logs klipper
* docker compose logs moonraker

Finally, run `scripts/get-info.sh` and share the generated **archive** here.

Tabaqui, 2025-05-01:
- docker compose logs klipper
- docker compose logs moonraker
Empty output for both.

[prind-info-01052025-131241.tar.gz](https://github.com/user-attachments/files/19998842/prind-info-01052025-131241.tar.gz)

mkuf, 2025-05-02:
Looking at the containers running on your system we can see that some containers are in the `Created` state instead of `Up`.  

```
CONTAINER ID   IMAGE                               COMMAND                  CREATED         STATUS                     PORTS                                 NAMES
046ae113f363   mkuf/moonraker:latest               "/opt/venv/bin/pytho…"   2 minutes ago   Created                                                          prind-moonraker-1
7c02b3c54c5e   mkuf/klipper:latest                 "/opt/venv/bin/pytho…"   2 minutes ago   Created                                                          prind-klipper-1
a27059d9caad   ghcr.io/fluidd-core/fluidd:latest   "/docker-entrypoint.…"   2 minutes ago   Up 2 minutes               80/tcp                                prind-fluidd-1
d77fd95dc652   mkuf/ustreamer:latest               "/opt/ustreamer --ho…"   2 minutes ago   Created                                                          prind-webcam-1
f51223acddc9   busybox:latest                      "chown -R 1000:1000 …"   2 minutes ago   Exited (0) 2 minutes ago                                         prind-init-1
07adfa9970a8   traefik:3.3                         "/entrypoint.sh --ac…"   2 minutes ago   Up 2 minutes               0.0.0.0:80->80/tcp, [::]:80->80/tcp   prind-traefik-1
```

For klipper and moonraker this might be due to the configured condition, but the webcam container should start right after it has been created, as it is not depending on any conditions to complete.  

I do assume that this is probably an issue with docker on your host system and not with prind.  
Seeing `Docker Root Dir: /home/tabaqui/docker/docker` leads me to believe that you're running docker in rootless mode, can you confirm this?  
If this is the case, this would explain why all containers that require access to the host system are stuck in the created state. 

Please also share the output of the following commands, which might show the reason why the containers are waiting. 

* docker inspect prind-klipper-1
* docker inspect prind-moonraker-1
* docker inspect prind-webcam-1q

Tabaqui, 2025-05-02:
I believe rootless is not the case here as I've never used it intentionally. I'm not sure of it but I've seen there's need to be some other special packages to install to be able to run docker in rootless mode. Till now I've only executed something like "install", "delete" package and `kill, prune, run, down`.

I was able to run moonraker and klipper as it says in your doc but then suddenly it started to get the opposite. I understand my explanation is not precise but I have no idea what I've been done wrong except maybe updating my packages that might have done something.

I've been playing around with all that images for some time which might be an issue(don't believe in it as it stopped working properly before). So I found that manual launching moonraker and klipper images starts python which correspondingly tells me that my printer.cfg and moonraker.cfg can't be read properly as they are not found at /opt/... or corrupted somehow. Still I don't know if it is important as I've just used `docker run -it <image>` to get that output.

Anyway here's the inspect command output:

[inspect_klipper.log](https://github.com/user-attachments/files/20012637/inspect_klipper.log)

[inspect_moonraker.log](https://github.com/user-attachments/files/20012639/inspect_moonraker.log)

[inspect_webcam.log](https://github.com/user-attachments/files/20012641/inspect_webcam.log)

mkuf, 2025-05-02:
The inspect output did not give any hints why those containers don't start.   

Check the docker daemon logs on your host, maybe they give a hint on why the containers remain in the created state.  
Also check the user under which the docker deamon runs (e.g. `ps aufx | grep docker`), if that is not root, your running docker in rootless mode. 

An example how to `run` the container can be found at https://github.com/mkuf/prind/tree/main/docker/klipper#run, the important bits here are the volume `/dev:/dev` and the privileged flag, which should not work when run in rootless mode

Given that you're on arch, any bleeding edge upgrades might have introduced some unwanted changes.

Tabaqui, 2025-05-02:
`groups $USER` shows `docker` and arch wiki for docker says:

> Warning: Anyone added to the docker group is root equivalent because they can use the docker run --privileged command to start containers with root privileges.

Also `ps aufx | grep docker`:
```
root        1598  0.0  0.2 3515344 75124 ?       Ssl  16:48   0:01 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock
root        7360  0.0  0.0 1671296 5708 ?        Sl   17:12   0:00  \_ /usr/bin/docker-proxy -proto tcp -host-ip 0.0.0.0 -host-port 80 -container-ip 172.18.0.4 -container-port 80 -use-listen-fd
root        7379  0.0  0.0 1745028 5740 ?        Sl   17:12   0:00  \_ /usr/bin/docker-proxy -proto tcp -host-ip :: -host-port 80 -container-ip 172.18.0.4 -container-port 80 -use-listen-fd
tabaqui     7720  0.0  0.0   7120  3900 pts/0    S+   17:18   0:00  |   \_ grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn --exclude-dir=.idea --exclude-dir=.tox --exclude-dir=.venv --exclude-dir=venv docker
root        7236  0.1  0.3 1440724 109240 ?      Ssl  17:12   0:00  \_ traefik traefik --accesslog --providers.docker=true --providers.docker.exposedbydefault=false --entrypoints.web.address=:80
```
so I think it's ok.

Yet I get while trying `docker compose ... up` :
```
level=error msg="Handler for POST /v1.49/containers/68a8ff8454b5f490a6d6b85aae51832ea4179e38b0fea7e95cf89b774c9e3660/start returned error: error gathering device information while adding custom device 
```

mkuf, 2025-05-02:
Right, that seems fine.  
I still haven't seen the docker home being something else than `/var/lib/docker` when running as root, but I'm mainly using debian based systems.  

That custom device from the daemon log may be the webcam, as this is the only service using the `device` directive.  

Are you at all able to start a container with the privileged flag and dev mounted as volume?
e.g.
```
docker run --rm -it --privileged -v /dev:/dev ubuntu:latest ls /dev
```

Tabaqui, 2025-05-02:
Yes, I'm able and yes, the issue was my camera. It seems I'm able to connect now.

Still I had to go through udev. Why? It is because in arch `uucp` group instead of `dialout` and gid is 14 in this case.

Thank you.

mkuf, 2025-05-02:
I'm glad to hear that this is resolved for you.  

Would you mind sharing the full udev rule you created?

Tabaqui, 2025-05-02:
Sure. Still I have got uncertainty
`ACTION=="add",SUBSYSTEM=="usb|tty",ATTRS{idVendor}=="1d50",ATTRS{idProduct}=="614e",MODE="0666",GROUP="14"`

I had to add MODE in it and usb subsystem seems useless I've just got no time to test it yet

closed 2025-05-07 by mkuf